dgiot/dgiot #113: The personal center can't be accessed

userliux, 2021-09-24:
**Environment**:

- dgiot version (e.g. `dgiot_ctl status`): 
- Hardware configuration (e.g. `lscpu`):
- OS (e.g. `cat /etc/os-release`):
- Kernel (e.g. `uname -a`):
- Erlang/OTP version (in case you build emqx from source code): 
- Others:

**What happened and what you expected to happen**:
The personal center can't be accessed

**How to reproduce it (as minimally and precisely as possible)**:

**Anything else we need to know?**:
[image]

h7ml, 2021-09-24:
hi. The reason why this link cannot jump to the personal center is because there is no corresponding menu configured.

h7ml, 2021-09-24:
[image]

done

closed 2021-09-24 by h7ml