dgiot/dgiot #215: Deploy on linux server failed

sevancheng, 2021-12-03:
**Environment**: CentOS 7.6
The deploy log as follows

[root@TestECS ~]# ls
[root@TestECS ~]# cd /
[root@TestECS /]# ls
bin  boot  dev  etc  home  lib  lib64  lost+found  media  mnt  opt  proc  root  run  sbin  srv  sys  tmp  usr  var
[root@TestECS /]# wget -q https://gitee.com/dgiiot/dgiot/raw/master/dgiot_install.sh  && sh dgiot_install.sh
osinfo: CentOS Linux
CentOS Linux 7 (Core)
cpe:/o:centos:centos:7
2021-12-03_13:39:36 970:  dgiot  single deploy  start
PING baidu.com (220.181.38.148) 56(84) bytes of data.
64 bytes from 220.181.38.148 (220.181.38.148): icmp_seq=1 ttl=50 time=30.7 ms
64 bytes from 220.181.38.148 (220.181.38.148): icmp_seq=2 ttl=50 time=30.7 ms

--- baidu.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 30.732/30.744/30.757/0.175 ms
2021-12-03_13:39:37 192:  lanip: 172.29.109.174
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    13  100    13    0     0     86      0 --:--:-- --:--:-- --:--:--    86
2021-12-03_13:39:37 180:  wlanip: 121.41.226.36
2021-12-03_13:40:10 366:  install_postgres
2021-12-03_13:40:11 373:  create postgres user and group 
Changing password for user postgres.
passwd: all authentication tokens updated successfully.
2021-12-03_13:40:13 352:  yum install postgres
2021-12-03_13:41:15 396:  configure postgres
2021-12-03_13:42:00 399:  make install postgres
2021-12-03_13:45:29 404:  make install pg_stat_statements 
2021-12-03_13:45:35 410:  build postgres sueccess
2021-12-03_13:45:36 426:  initdb postgres success
2021-12-03_13:45:36 472:  /data/dgiot/dgiot_pg_writer/data/postgresql.conf
2021-12-03_13:45:36 230:  build /lib/systemd/system/dgiot_pg_writer.service
2021-12-03_13:45:37 269:  systemctl start dgiot_pg_writer
2021-12-03_13:45:39 484:  deploy postgres success
2021-12-03_13:45:53 552:  create /data/dgiot/dgiot_parse_server/script/.env success
2021-12-03_13:45:54 561:  install parse_server success
2021-12-03_13:46:16 576:  install install_dgiot_redis success
2021-12-03_13:46:16 316:  build /lib/systemd/system/dgiot_redis.service
2021-12-03_13:46:17 345:  systemctl start dgiot_redis
2021-12-03_13:46:17 316:  build /lib/systemd/system/dgiot_parse_server.service
2021-12-03_13:46:18 345:  systemctl start dgiot_parse_server
2021-12-03_13:46:19 836:  pg_stat_statements has installed
2021-12-03_13:46:20 316:  build /lib/systemd/system/postgres_exporter.service
2021-12-03_13:46:20 345:  systemctl start postgres_exporter
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
2021-12-03_13:46:44 316:  build /lib/systemd/system/dgiot_tdengine_mqtt.service
2021-12-03_13:46:45 345:  systemctl start dgiot_tdengine_mqtt
2021-12-03_13:46:46 275:  build /lib/systemd/system/gofastdfs.service
2021-12-03_13:46:46 310:  systemctl start gofastdfs
mkdir: cannot create directory ‘/data/dgiot/go_fastdfs/files’: File exists
[root@TestECS /]#

dawnwinterLiu, 2021-12-03:
The script has been updated, delete the dgiot folder in the data directory, and reinstall

sevancheng, 2021-12-03:
resolved

closed 2021-12-03 by sevancheng