dgiot/dgiot #657: make run无法执行完成

ljm000, 2023-07-03:
按照github主页要求，拉取完代码后
[image]

make run无法执行完成，最后提示以下错误
<pre>
===> Compiling c_src/iconverl.c
===> Linking /root/dgiot/dgiot/apps/dgiot/priv/iconv.so
===> Writing /root/dgiot/dgiot/apps/emqx_exhook/src/emqx_exhook_pb.erl
===> Writing /root/dgiot/dgiot/apps/emqx_exhook/src/emqx_exhook_v_1_hook_provider_client.erl
===> Writing /root/dgiot/dgiot/apps/emqx_exhook/src/emqx_exhook_v_1_hook_provider_bhvr.erl
===> Writing /root/dgiot/dgiot/apps/dgiot_dlink/src/grpc/dgiot_dlink_client.erl (forcibly overwriting)
===> Writing /root/dgiot/dgiot/apps/dgiot_dlink/src/grpc/dgiot_dlink_bhvr.erl (forcibly overwriting)
===> Analyzing applications...
===> Compiling emqx
===> Compiling dgiot
===> Compiling dgiot_api
===> Compiling dgiot_bacnet
===> Compiling dgiot_bamis
===> Compiling dgiot_bridge
===> Compiling dgiot_device
===> Compiling dgiot_dlink
===> Compiling dgiot_evidence
===> Compiling dgiot_factory
===> Compiling dgiot_ffmpeg
===> Compiling dgiot_gb26875
===> Compiling dgiot_hjt212
===> Compiling dgiot_http
===> Compiling dgiot_location
===> Compiling dgiot_meter
===> Compiling dgiot_modbus
===> Compiling dgiot_mysql
===> Compiling dgiot_opc
===> Compiling dgiot_parse
===> Compiling dgiot_printer
===> Compiling dgiot_task
===> Compiling dgiot_tdengine
===> Compiling dgiot_topo
===> Compiling emqx_auth_mnesia
===> Compiling emqx_bridge_mqtt
===> Compiling emqx_exhook
===> Compiling emqx_management
===> Compiling emqx_plugin_libs
===> Compiling emqx_prometheus
===> Compiling emqx_psk_file
===> Compiling emqx_recon
===> Compiling emqx_retainer
===> Compiling emqx_rule_engine
===> Compiling apps/emqx_rule_engine/src/emqx_rule_engine_jwt_worker.erl failed
emqx_rule_engine_jwt_worker.erl:87: function alias/1 undefined

make: *** [Makefile:101: emqx] Error 1
</pre>

ljm000, 2023-07-03:
[image]

lsxredrain, 2023-08-08:
有没有安装erlang环境